makers-for-life/makair-firmware #6: Compiler error with HARDWARE_VERSION 2

ilyautov, 2020-05-09:
Hi, when i try to compile firmware i getting next compiler errors:
`/Users/ilyautov/Library/Arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /var/folders/zh/zv6q8qk55fzfqn2wynsx_r4w0000gn/T/arduino-sketch-3F591D5A8DAFBDBDD367C76F7C00AB59/sketch/keyboard.cpp.o: in function `initKeyboard()':
keyboard.cpp:(.text._Z12initKeyboardv+0x14): undefined reference to `AnalogButtons::add(Button)'
/Users/ilyautov/Library/Arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: keyboard.cpp:(.text._Z12initKeyboardv+0x26): undefined reference to `AnalogButtons::add(Button)'
/Users/ilyautov/Library/Arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: keyboard.cpp:(.text._Z12initKeyboardv+0x38): undefined reference to `AnalogButtons::add(Button)'
/Users/ilyautov/Library/Arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: keyboard.cpp:(.text._Z12initKeyboardv+0x4a): undefined reference to `AnalogButtons::add(Button)'
/Users/ilyautov/Library/Arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: keyboard.cpp:(.text._Z12initKeyboardv+0x5c): undefined reference to `AnalogButtons::add(Button)'
/Users/ilyautov/Library/Arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /var/folders/zh/zv6q8qk55fzfqn2wynsx_r4w0000gn/T/arduino-sketch-3F591D5A8DAFBDBDD367C76F7C00AB59/sketch/keyboard.cpp.o:keyboard.cpp:(.text._Z12initKeyboardv+0x6e): more undefined references to `AnalogButtons::add(Button)' follow
/Users/ilyautov/Library/Arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /var/folders/zh/zv6q8qk55fzfqn2wynsx_r4w0000gn/T/arduino-sketch-3F591D5A8DAFBDBDD367C76F7C00AB59/sketch/keyboard.cpp.o: in function `keyboardLoop()':
keyboard.cpp:(.text._Z12keyboardLoopv+0x4): undefined reference to `AnalogButtons::check()'
/Users/ilyautov/Library/Arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /var/folders/zh/zv6q8qk55fzfqn2wynsx_r4w0000gn/T/arduino-sketch-3F591D5A8DAFBDBDD367C76F7C00AB59/sketch/keyboard.cpp.o: in function `_GLOBAL__sub_I__Z22onPeakPressureIncreasev':
keyboard.cpp:(.text.startup._GLOBAL__sub_I__Z22onPeakPressureIncreasev+0x10): undefined reference to `AnalogButtons::AnalogButtons(unsigned char, unsigned char, unsigned short, unsigned char)'
/Users/ilyautov/Library/Arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: keyboard.cpp:(.text.startup._GLOBAL__sub_I__Z22onPeakPressureIncreasev+0x4c): undefined reference to `Button::Button(unsigned short, void (*)(), void (*)(), unsigned short, unsigned short)'
/Users/ilyautov/Library/Arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: keyboard.cpp:(.text.startup._GLOBAL__sub_I__Z22onPeakPressureIncreasev+0x5e): undefined reference to `Button::Button(unsigned short, void (*)(), void (*)(), unsigned short, unsigned short)'
/Users/ilyautov/Library/Arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: keyboard.cpp:(.text.startup._GLOBAL__sub_I__Z22onPeakPressureIncreasev+0x70): undefined reference to `Button::Button(unsigned short, void (*)(), void (*)(), unsigned short, unsigned short)'
/Users/ilyautov/Library/Arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: keyboard.cpp:(.text.startup._GLOBAL__sub_I__Z22onPeakPressureIncreasev+0x82): undefined reference to `Button::Button(unsigned short, void (*)(), void (*)(), unsigned short, unsigned short)'
/Users/ilyautov/Library/Arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: keyboard.cpp:(.text.startup._GLOBAL__sub_I__Z22onPeakPressureIncreasev+0x94): undefined reference to `Button::Button(unsigned short, void (*)(), void (*)(), unsigned short, unsigned short)'
/Users/ilyautov/Library/Arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /var/folders/zh/zv6q8qk55fzfqn2wynsx_r4w0000gn/T/arduino-sketch-3F591D5A8DAFBDBDD367C76F7C00AB59/sketch/keyboard.cpp.o:keyboard.cpp:(.text.startup._GLOBAL__sub_I__Z22onPeakPressureIncreasev+0xa6): more undefined references to `Button::Button(unsigned short, void (*)(), void (*)(), unsigned short, unsigned short)' follow
collect2: error: ld returned 1 exit status
Using library IWatchdog at version 1.0.0 in folder: /Users/ilyautov/Library/Arduino15/packages/STM32/hardware/stm32/1.8.0/libraries/IWatchdog 
Using library LiquidCrystal at version 1.0.7 in folder: /Users/ilyautov/Documents/Arduino/libraries/LiquidCrystal 
Using library SrcWrapper at version 1.0.1 in folder: /Users/ilyautov/Library/Arduino15/packages/STM32/hardware/stm32/1.8.0/libraries/SrcWrapper 
Using library OneButton at version 1.5.0 in folder: /Users/ilyautov/Documents/Arduino/libraries/OneButton 
Error during build: exit status 1`
if i'm adding `#include <AnalogButtons.h>` for all versions of hardware all working well, can you help me?

ilyautov, 2020-05-09:
if i'm adding`#include <AnalogButtons.h>` for all versions of hardware all working well, can you help me?

dsferruzza, 2020-05-09:
Hi!

I don't seem to be able to reproduce this... Could you tell me which commit are you trying to build? Also what is the content of `includes/config.h` in your case? (I assume that except for `includes/config.h` you did not modify anything?)

ilyautov, 2020-05-09:
Hello, @dsferruzza. 
Thank you for your response.
i'm trying to build that in mac os, previously i made step by step installation, and now i'm using commit  a4d72438765125b28c9a2b956235f1ac849e64f0(https://github.com/makers-for-life/makair-firmware/commit/a4d72438765125b28c9a2b956235f1ac849e64f0)
i'm modified only HARDWARE_VERSION.

so my config file
`
/******************************************************************************
 * @author Makers For Life
 * @copyright Copyright (c) 2020 Makers For Life
 * @file config.h
 * @brief Main configuration
 *****************************************************************************/

#pragma once

// Available modes
#define MODE_PROD 1              ///< Ventilator mode
#define MODE_QUALIFICATION 2     ///< Test electrical wiring mode
#define MODE_INTEGRATION_TEST 3  ///< Test integration mode

/// Defines the current mode
#define MODE MODE_PROD

/**
 * Activates debug traces
 *
 * When DEBUG = 1, additional code is added to send debug traces using serial
 */
#define DEBUG 1

/**
 * Activates pressure sensor simulation
 *
 * When SIMULATION = 1, the current pressure value will not be read from the sensor but computed by
 * a basic and deterministic model
 */
#define SIMULATION 0

// Available pneumatic versions
#define PHW_PIGGY 0
#define PHW_CHU 1
#define PHW_FAULHABER 2

/// Defines which preset to use for controlling pressure
#define PNEUMATIC_HARDWARE_VERSION PHW_CHU

// Available valves
#define VT_SERVO_V1 0
#define VT_EMERSON_ASCO 1
#define VT_FAULHABER 2

// Defines which valves are fitted
#define VALVE_TYPE VT_SERVO_V1

/**
 * Defines which hardware preset to use
 *
 * This changes which hardware is used, to which pins it is connected, as well as various parameters
 */
#define HARDWARE_VERSION 2
`

dsferruzza, 2020-05-09:
That is weird because with `HARDWARE_VERSION 2` you should not need the AnalogButton lib at all... (The error you submitted is really hard to read if you do not wrap it with triple back-ticks (```).)

We sometimes had errors because incremental compilation was doing weird things. This can be fixed by deleting the temporary directory produced during compilation (the one that has `arduino-sketch` in its name; see error message).

dsferruzza, 2020-05-09:
Also, here is how the CI installs everything: https://github.com/makers-for-life/makair-firmware/blob/master/.github/workflows/firmware-release.yml#L17-L25 Maybe something important differs from your current setup?

ilyautov, 2020-05-09:
@dsferruzza sorry, i fixed markdown. ok i will try to delete directory.
so i made same installation 
```
arduino-cli config init --additional-urls https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
          arduino-cli config dump
          arduino-cli core update-index
          arduino-cli core install STM32:stm32
          arduino-cli lib install LiquidCrystal
          arduino-cli lib install "Analog Buttons"
          arduino-cli lib install OneButton
```

ilyautov, 2020-05-09:
i will try to make clear installation in container

ilyautov, 2020-05-10:
Hi @dsferruzza.
I resolved this issue by using docker.
How i can contribute that solution?

dsferruzza, 2020-05-16:
@ilyautov Sorry I took so long to answer, I add to work on other projects. I believe we can close this issue and you took contact with @valeriansaliou, right?

closed 2020-10-17 by dsferruzza