makers-for-life/makair-firmware #16: Telemetry's DataSnapshot.battery_level is reported as 27V, and not 27xV (more precise)

valeriansaliou, 2020-05-16:
Would it be possible that the firmware's telemetry system report the lead acid battery voltage with an added decimal of precision, for more precise % calculations after we characterized the discharge profile of our batteries?

As the current battery_level is encoded on an u8, we'd need to apply an offset in order to encode "270", eg. consider zero as 20V, and start from there, which means that "270" would be encoded as "70" on the same u8, which equals to "27.0V".

dsferruzza, 2020-10-17:
To ensure compatibility and simplicity, the best option is probably to add a new u16 field "accurate_battery_level"

baptistejamin, 2020-12-30:
It has been implemented

closed 2020-12-30 by baptistejamin