makers-for-life/makair-firmware #25: Add possibility to go to PEP 0

waxzce, 2020-05-26:


Benhalor, 2020-05-26:
I do it

jabby, 2020-05-26:
What is the goal of set a PEEP below alarm threshold?

Benhalor, 2020-05-26:
The min peep setting is now 5 cmH2O.
It should be 0 cm H2O, because some medical cases require this kind of peep

jabby, 2020-05-26:
Ok I understand. How alarms should work for this kind of settings?

See the following alarms: 
* https://github.com/makers-for-life/makair-firmware/blob/master/srcs/alarm_controller.cpp#L34-L39
* https://github.com/makers-for-life/makair-firmware/blob/master/srcs/alarm_controller.cpp#L54-L59
* https://github.com/makers-for-life/makair-firmware/blob/master/srcs/alarm_controller.cpp#L104-L109

Benhalor, 2020-10-17:
@Benhalor should ask the regulatory team if the feature is still mandatory. And how to deal with the alarms issue pointed out by @jabby