makers-for-life/makair-firmware #68: Rpi watchdog

pi-r-p, 2020-11-05:
Idea : implement a heartbeat message from Rpi to STM32. 
If the STM32 do not see the hearbeat during one minute, it resets the Rpi the hardware way. The HW v3 can control Rpi power lines.

valeriansaliou, 2020-11-05:
Will be implemented tomorrow in the UI.

dsferruzza, 2020-11-07:
- The control protocol was updated to support heartbeat messages
- If one of these messages has the special value `43690`, the MCU must disable the Rpi watchdog until it is restarted (otherwise the value is ignored)
- A command was added to the telemetry CLI to send this special heartbeat once (in case the UI is down and we want to work on the RPi): just run `./scripts/run.sh disable-rpi-watchdog --port [PORT NAME/ADDRESS]` from the telemetry directory

---

@Benhalor I prepared `serial_control.cpp`. I suggest:

- a class to handle this feature and its state
- methods for:
  - receiving a heartbeat
  - checking if RPi must be restarted (if it is restarted, we need to wait for a valid heartbeat before enabling the watchdog back)
  - disabling the watchdog until restart
- a macro in `config.h` to force disabling the watchdog

closed 2020-11-14 by dsferruzza